jgromes/RadioLib #1480: "Join failed: -1116" (DevNonce has already been used) on LoRaWAN_ESP32.ino (radiolib-persistence) with TTGO LoRa32 After Power Cycle

ElectronicallyE, 2025-04-08:
Hello,

This is a follow-on from a post I made on the ChripStack Forum (the network server I am using):
https://forum.chirpstack.io/t/device-on-otaa-ttgo-lora32-v1-not-connecting-unless-flush-otaa-device-nonces-invalid-devnonce/24273

Following the advice of ChirpStack Forum user bns (@StevenCellist), I am using the LoRaWAN_ESP32.ino example:
https://github.com/radiolib-org/radiolib-persistence/blob/main/examples/LoRaWAN_ESP32/LoRaWAN_ESP32.ino

I acknowledge that the example is in a different repository to this one, however as it only includes example sketches and uses this library, I though I would share it here.

My code is as follows:
**LoRaWAN_ESP32.ino**
```
/*

This demonstrates how to save the join information in to permanent memory
so that if the power fails, batteries run out or are changed, the rejoin
is more efficient & happens sooner due to the way that LoRaWAN secures
the join process - see the wiki for more details.

This is typically useful for devices that need more power than a battery
driven sensor - something like a air quality monitor or GPS based device that
is likely to use up it's power source resulting in loss of the session.

The relevant code is flagged with a ##### comment

Saving the entire session is possible but not demonstrated here - it has
implications for flash wearing and complications with which parts of the 
session may have changed after an uplink. So it is assumed that the device
is going in to deep-sleep, as below, between normal uplinks.

Once you understand what happens, feel free to delete the comments and 
Serial.prints - we promise the final result isn't that many lines.

*/

#if !defined(ESP32)
  #pragma error ("This is not the example your device is looking for - ESP32 only")
#endif

// ##### load the ESP32 preferences facilites
#include <Preferences.h>
Preferences store;

// LoRaWAN config, credentials & pinmap
#include "config.h" 

#include <RadioLib.h>

// utilities & vars to support ESP32 deep-sleep. The RTC_DATA_ATTR attribute
// puts these in to the RTC memory which is preserved during deep-sleep
RTC_DATA_ATTR uint16_t bootCount = 0;
RTC_DATA_ATTR uint16_t bootCountSinceUnsuccessfulJoin = 0;
RTC_DATA_ATTR uint8_t LWsession[RADIOLIB_LORAWAN_SESSION_BUF_SIZE];

// abbreviated version from the Arduino-ESP32 package, see
// https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/deepsleep.html
// for the complete set of options
void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason = esp_sleep_get_wakeup_cause();
  if (wakeup_reason == ESP_SLEEP_WAKEUP_TIMER) {
    Serial.println(F("Wake from sleep"));
  } else {
    Serial.print(F("Wake not caused by deep sleep: "));
    Serial.println(wakeup_reason);
  }

  Serial.print(F("Boot count: "));
  Serial.println(++bootCount);      // increment before printing
}
[... 647 lines cut ...]
19:02:45.458 -> RLB_PRO: Closing Rx2 window
19:02:45.458 -> FCntUp: 0
19:02:45.458 -> Sleeping
19:02:45.458 -> 
```

**Output from "Events" of Device in ChirpStack for Serial Output 2**

[image]

[image]

**Screenshot From Console Output of Raspberry Pi Running ChirpStack Showing Error**
[image]

I acknowledge that in the forum post, @StevenCellist said he is preoccupied with the many responsibilities of life, but I just wanted to document my findings.

Objective: To figure out why the DevNonces are being reused if they are being saved.

**Additional info:**
 - MCU: TTGO LoRa32 V1 (choose TTGO LoRa32-OLED)
 - Link to Arduino core: https://github.com/espressif/arduino-esp32
 - Wireless module type: SX1276
 - Arduino IDE version: 2.3.4
 - Library version: 7.1.2

StevenCellist, 2025-04-08:
Heh, I think I have the idea.

I bet your device restarts automatically when you open the serial monitor?
So, what then happens: you plug in your ESP32, it boots, it transmits a JoinRequest, you open the serial monitor, it resets, it transmits a JoinRequest. But as it never got to complete its previous JR/JA cycle, the DevNonces weren't updated so the new JR is transmitted using the same DevNonce.
The solution is to add a delay in the begin of the `setup()`, or otherwise hold down the PRG/USER/whatever button other than RST when connecting your device's power, which means it waits for a restart/download.

ElectronicallyE, 2025-04-08:
Before connecting the device:
[image]

After plugging it in:
[image]

StevenCellist, 2025-04-08:
Something sus. See the entries 09:31:28.487 and .488: two JoinRequests back-to-back. The first gets validated and sets DevNonce = 30, the second one subsequently results in an ERROR because DevNonce = 30 is already used. Something up with your Chirpstack? Or overloading the antenna because they're very close to each other (no reports of RSSI).

ElectronicallyE, 2025-04-08:
I have just added the output for the first serial output to my original post.

Copy below as well:

[image]

StevenCellist, 2025-04-08:
Exactly the same problem: JoinRequest goes twice into the system within 3ms. I guess there's a configuration mistake on your side that injects traffic twice into the system or something like that. This is definitely not due to RadioLib or your device transmitting twice - 3ms is way too tight for that. it's much more typical for internet traffic.

Please use Markdown-details next time to hide stuff - so many pictures and so many logs. Can't cope on smaller display.

HeadBoffin, 2025-04-08:
> I acknowledge that the example is in a different repository to this one, however as it only includes example sketches and uses this library, I though I would share it here.

Erm, thanks for that, wondering if you appreciate that the two repros are in fact looked after by the same group of people?

It may be worth your time to join up all the dots across the various forums so that, well, you don't need to explain it all over again to the same people.

To save further time, no point posting a photo of a screen - we can't parse it and we can barely read it.

ElectronicallyE, 2025-04-08:
Hi @HeadBoffin,

Thank you for your comment. I acknowledge that both repos are maintained by the same people. I found that far more matter have been raised for the library itself rather than the examples. If you would rather I had raised it in the other repository, I am happy to do so in future if there are any further issues.

I will also be happy to share logs in future instead of a photo of a screen by ssh'ing into the Pi instead of running it standalone (which I was this evening).

ElectronicallyE, 2025-04-08:
@StevenCellist,

By seeing the two join requests within 3 ms of one another, are you concluding that they are duplicates and have the same devnonce? Is there anywhere which is explicitly calling out they are the same?

Do you believe this is a ChripStack issue or the gateway (I'm using a Laird RG191)? If so, are there any potential places I shall look?

I have also added "details" markup to my original post as requested.

StevenCellist, 2025-04-08:
I believe you have enough pointers to do further research. Re-read some messages here, look at the CS logs, especially the very obvious ERROR that is displayed, along with the preceding lines that lead up to it. I do not believe that the issue is in RadioLib but rather in your setup / gateway <> Chirpstack. We cannot help you further here, sorry. Now if you were to sponsor my student project as I am helping your business case, I may be more willing. But closing this now & here.

closed 2025-04-08 by StevenCellist

ElectronicallyE, 2025-04-08:
Thank you for all your help and insight @StevenCellist, I really appreciate it.

HeadBoffin, 2025-04-08:
> If you would rather I had raised it in the other repository, I am happy to do so in future if there are any further issues.

No, I'd rather not have Google find a whole wodge of code in an issue that's a copy from another repro. People tend to CopyNPasta the first thing they find. Referencing it OK, copy not so much.