jgromes/RadioLib #1484: CC1101: packet discarded on chip are delivered to caller

jcrespoc, 2025-04-16:
Node 1 sending packets to node address 0x03.
Node 2 with address 0x02 receive garbage.

CC1101::receive waits for !Irq (GDO0 asserted) + Irq (GDO0 de-asserted) to read the packet.
Inside, CC1101::readData assumes that data exists in the FIFO without verifying it.

If the packet is discarded, GDO0 is de-asserted as well.

`RADIOLIB_CC1101_GDOX_SYNC_WORD_SENT_OR_PKT_RECEIVED = 0x06` 
Asserts: when sync word has been received
de-asserts: at the end of the packet, **when a packet is discarded due to address or maximum length filtering**, or when the radio enters the RXFIFO_OVERFLOW state.

An arbitrary number, which is assumed to be the length, is read from an empty FIFO, and from there another bunch of arbitrary data.
Finally, one is read that is assumed to contain the CRC, and if bit7 is raised, the packet is considered correct.

<details><summary>Debug mode output</summary>
<p>

```
[CC1101] Initializing ... success!
[CC1101] Setting BitRate ... success!
[CC1101] Setting node address (0x02)... success!
[CC1101] Setting variable packet length mode ... success!
[CC1101] Waiting for incoming transmission ... success!
received data:
0x00, 0x00, 0x00, 0xac, 0x30, 0x00, 0x82, 0x30, // ....0..0
[CC1101] RSSI:          -44.50 dBm
[CC1101] LQI:           126
[CC1101] Waiting for incoming transmission ... success!
received data:
0x00, 0x00, 0x00, 0xac, 0x30, 0x00, 0x82, 0x30, // ....0..0
[CC1101] RSSI:          -44.50 dBm
[CC1101] LQI:           126

Again with some debug output added to RadioLib:

[CC1101] Setting BitRate ... success!
[CC1101] Setting node address (0x02)... success!
[CC1101] Setting variable packet length mode ... success!

[CC1101] Waiting for incoming transmission ... 
[  1985][D][CC1101_Receive_Address.ino:110] operator()(): GDO0 triggered 0
[  1999][D][CC1101_Receive_Address.ino:110] operator()(): GDO0 triggered 1
[  2009][D][CC1101.cpp:828] getPacketLength(): bytes in fifo: 0
[  2015][D][CC1101.cpp:830] getPacketLength(): packet length 0
[  2021][D][CC1101.cpp:384] readData(): length 0
[  2025][D][CC1101.cpp:396] readData(): addr 57
success!
received data:
0x8a, 0xc9, 0x3f, 0xff, 0xff, 0x3f, 0xb3, 0x00, // ..?..?..
[CC1101] RSSI:          -75.00 dBm
[CC1101] LQI:           50
[CC1101] Waiting for incoming transmission ... 
[  3094][D][CC1101_Receive_Address.ino:110] operator()(): GDO0 triggered 0
[  3108][D][CC1101_Receive_Address.ino:110] operator()(): GDO0 triggered 1
[  3118][D][CC1101.cpp:828] getPacketLength(): bytes in fifo: 0
[  3124][D][CC1101.cpp:830] getPacketLength(): packet length 190
[  3130][D][CC1101.cpp:384] readData(): length 190
[  3134][D][CC1101.cpp:389] readData(): new length 8
[  3139][D][CC1101.cpp:396] readData(): addr 57
0xfe, 0xb2, 0xbe, 0xa0, 0x3f, 0x14, 0xb7, 0xfe, // ....?...
CRC error!

```

</p>
</details> 

<details><summary>Slightly modified CC1101_Receive_Address.ino sketch"</summary>
<p>

```c++
#ifdef EXAMPLERX

/*
  RadioLib CC1101 Receive with Address Example

  This example receives packets using CC1101 FSK radio
  module. Packets can have 1-byte address of the
  destination node. After setting node address, this node
  will automatically filter out any packets that do not
  contain either node address or broadcast addresses.

  For default module settings, see the wiki page
  https://github.com/jgromes/RadioLib/wiki/Default-configuration#cc1101

  For full API reference, see the GitHub Pages
  https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// CC1101 has the following connections:
// CS pin:    10
// GDO0 pin:  9
// RST pin:   unused
// GDO2 pin:  46
CC1101 radio = new Module(10, 9, RADIOLIB_NC, 46);

#define TXADDR 0x01
#define RXADDR 0x02

// or detect the pinout automatically using RadioBoards
// https://github.com/radiolib-org/RadioBoards
/*
#define RADIO_BOARD_AUTO
#include <RadioBoards.h>
Radio radio = new RadioModule();
*/

void setup() {
  Serial.begin(115200);
  delay(500);

  pinMode(46, INPUT_PULLUP);
  pinMode(9, INPUT_PULLUP);

  // initialize CC1101 with default settings
  Serial.print(F("[CC1101] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }

  Serial.printf(F("[CC1101] Setting BitRate ... "));
  state = radio.setBitRate(1.2f);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }

  // set node address
  // NOTE: Calling this method will automatically enable
  //       address filtering. CC1101 also allows to set
  //       number of broadcast address (0/1/2).
  //       The following sets one broadcast address 0x00.
  //       When setting two broadcast addresses, 0x00 and
  //       0xFF will be used.
  Serial.printf(F("[CC1101] Setting node address (0x%02x)... "), RXADDR);
  state = radio.setNodeAddress(RXADDR, 1);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }

  Serial.print(F("[CC1101] Setting variable packet length mode ... "));
  state = radio.variablePacketLengthMode();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }

  // address filtering can also be disabled
  // NOTE: Calling this method will also erase previously
  //       set node address
  /*
    Serial.print(F("[CC1101] Disabling address filtering ... "));
    state == radio.disableAddressFiltering();
    if(state == RADIOLIB_ERR_NONE) {
      Serial.println(F("success!"));
    } else {
      Serial.print(F("failed, code "));
      Serial.println(state);
      while (true) { delay(10); }
    }
  */
 radio.setGdo0Action([]{
  log_d("GDO0 triggered %d", digitalRead(9));
 }, CHANGE);
}

void loop() {
  Serial.print(F("[CC1101] Waiting for incoming transmission ... "));
  // you can receive data as an Arduino String
  //String str;
  //int state = radio.receive(str);

  // you can also receive data as byte array
  
    byte byteArr[RADIOLIB_CC1101_MAX_PACKET_LENGTH];
    int state = radio.receive(byteArr, 8);
  

  if (state == RADIOLIB_ERR_NONE) {
    // packet was successfully received
    Serial.println(F("success!"));

    // print the data of the packet
    //Serial.print(F("[CC1101] Data:\t\t"));
    //Serial.println((const char*) byteArr);

    Serial.println("received data:");
    log_buf_i(byteArr, 8);

    //Serial.println("expected data:");
    //byte byteArrSent[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    //log_buf_i(byteArrSent, 8);

    // print RSSI (Received Signal Strength Indicator)
    // of the last received packet
    Serial.print(F("[CC1101] RSSI:\t\t"));
    Serial.print(radio.getRSSI());
    Serial.println(F(" dBm"));

    // print LQI (Link Quality Indicator)
    // of the last received packet, lower is better
    Serial.print(F("[CC1101] LQI:\t\t"));
    Serial.println(radio.getLQI());

  } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }
}

#endif
```

</p>
</details> 

**Expected behavior**
Discarded packets should be... discarded

**Additional info (please complete):**
 - MCU: ESP32-S3

lyusupov, 2025-04-17:
Not sure that PR gives a positive effect for my use case...
Amount of Rx packets registered is still greater than packets actually sent out

[image]

jgromes, 2025-04-17:
@jcrespoc this is indeed a bug, however, I disagree with the solution proposed in #1486. It adds an external loop that will restart reception in blocking mode, until a valid packet is received (or reception times out). This doesn't solve interrupt-driven reception, which will still report data for discarded packets.

To me, the clean solution would be either to filter out these "discarded packet" GDO signals by using a different interrupt source from the radio (there is one that asserts at the end of packet when CRC check passes; but I'm not sure how that would behave with CRC disabled). Alternatively, we could perform a "data valid" check in `readData` to make sure there is actually valid stuff in the Rx FIFO, and if not, return some error code. What do you think?

jcrespoc, 2025-04-17:
> [@jcrespoc](https://github.com/jcrespoc) this is indeed a bug, however, I disagree with the solution proposed in [#1486](https://github.com/jgromes/RadioLib/pull/1486). It adds an external loop that [...]

Regarding the outer loop: this actually happens when a packet is dropped, either because it is for another node, or because a wrong length was read... I don't see why receive should return. It makes more sense to me to keep waiting until the reception timer expires.

Regarding interrupt-driven reception (CC1101_Receive_Interrupt.ino): my getPacketLength returns 0 if there is no length to read in FIFO. getData could returns error but calling the client function to throwing an error is ugly.

If interrupt source based on 7 (Asserts when a packet has been received with CRC OK. De-asserts when the first byte is read from the RX FIFO) might work regardless of CRC. Who knows. 
In my opinion, the interrupt-based solution should process the interrupt internally before calling the function provided to setPacketReceived. So simple.

I'm finishing new receive and transmit operations of unlimited length (infinite length mode down to the last fragment).
It also remains in reception mode (It's not valid for me the time it takes to get back to reception through Idle) and reports packet available to a callback. Let me know if you're interested.

Thanks

jgromes, 2025-04-18:
> In my opinion, the interrupt-based solution should process the interrupt internally before calling the function provided to setPacketReceived. So simple.

Unfortunately, it is anything but simple. Processing the interrupt internally will require performing an SPI transaction. However, there is no guarantee the platform RadioLib is running on can perform SPI transactions in ISR context. That's why in all examples in all radios, the only thing that happens in an ISR is raising a flag to processed later.

I will try to test whether the other interrupt works, since that would resolve all of these problems.

> I'm finishing new receive and transmit operations of unlimited length

That would be interesting - we actually already have interface for that, it's called the [Stream mode](https://github.com/jgromes/RadioLib/tree/master/examples/Stream). You can see its implementation in the SX127x class for example, in `setFifoEmptyAction`, `fifoAdd`, `fifoGet` etc. Feel free to open a new issue/PR on this topic.

closed 2025-07-06 by jgromes