jgromes/RadioLib #1487: Uploading Firmware / Corruption always in DFU

gc9n, 2025-04-16:
hello there , i am not sure if its a bug or not , but while i was updating the firmare accidentally the VCC unplugged.
on the next power cycle i was expecting to repeat the proccess succesfull but the result was that the 
CHIP was recognised as DFU 
#define RADIOLIB_LR11X0_DEVICE_BOOT                             (0xDFUL << 0)   //  7     0                bootloader mode

is there any way to save this piece of hardware? 

[image]

jgromes, 2025-04-16:
I don't think it's bricked, it just looks like the device remained in bootloader mode after teh pwoer cycle, and RadioLib doesn't accept that as a valid device ID since I assumed that FW update would be done without resetting the device or calling `begin` again. However, there's nothing preventing us from accepting the ID and letting the startup run. It will most likely fail elsewhere in `begin` since basic functionality (like setting the frequency) is probably not available in bootloader mode, but you can just discard the error code and update the firmware again.

As a general rule of thumb (though I'm sure it was accidental), for firmware updates you really need to make sure the setup is stable and uninterrupted ;)

jgromes, 2025-05-01:
Seeings as there is no further activity, I take it this was resolved by the mentioned commit.

closed 2025-05-01 by jgromes