jgromes/RadioLib #1488: Can't get any example to work on Lilygo TS-S3 with sx1280PA chip

Nikita-Kononov, 2025-04-17:
I have 2 Lilygo T3-S3 2.4GHz modules for a wireless telemetry project. I cant get any of the examples to work however. Both modules function fine as ESP32 boards (can run different sensors for example) but i cant get them to communicate with each other. Using the Lilygo examples both modules initialize ok, the receiver starts listening ok but after the transmitter sends the first packet i get no confirmation in serial of it being sent successfully. I also tried StuartProjects SX12XX library with the same results. The transmitter tries to send the packet but it never does. The strange thing is that no error codes ever get triggered. I modified the code to print the status to serial and it shows 0 every time. No timeout is triggered either. Both sender and receiver are able to set the frequency correctly as well and all settings are the same for both modules. I tried switching the modules around but achieved the same result. I'm new to this and I feel like I'm making a stupid mistake somewhere but I can't find it and need help.

<details><summary>Sketch that is causing the module fail</summary>
<p>

As the code is too long to post here: this is the code that i used from lilygo's library. it uses the RadioLib library. The only modifications I made to the code is: uncomment my board type in the utilities.h file and changed the pin numbers according to the schematic I listed below.
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/Transmit_Interrupt

These are the StuartProjects examples that i used:
https://github.com/StuartsProjects/SX12XX-LoRa/tree/master/examples/SX128x_examples/ESP32
Only modification made was the pin numbers in the settings.h file according to the schematic below. 

</p>
</details> 

**Hardware setup**
the schematic from liligo that i used for the pins:

[image]

[image]

this is the serial output for the transmitter:

[image]

this is the serial output for the receiver:

[image]

<details><summary>SPI Debug mode output</summary>
<p>

RLB_SPI: CMDW	80	
RLB_SPI: SI		0	
RLB_SPI: SO	FF	FF	
RLB_SPI: CMDW	80	
RLB_SPI: SI		0	
RLB_SPI: SO	FF	43	
RLB_SPI: CMDW	80	
RLB_SPI: SI		0	
RLB_SPI: SO	45	43	
RLB_SPI: CMDW	8F	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	43	43	
RLB_SPI: CMDW	8A	
RLB_SPI: SI		1	
RLB_SPI: SO	45	43	
RLB_SPI: CMDW	88	
RLB_SPI: SI		60	
RLB_SPI: SO	45	43	
RLB_SPI: CMDW	96	
RLB_SPI: SI		1	
RLB_SPI: SO	45	43	
RLB_SPI: CMDW	86	
RLB_SPI: SI		B8	9D	89	
RLB_SPI: SO	45	43	43	43	
RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	8B	
RLB_SPI: SI		90	18	3	
RLB_SPI: SO	45	43	43	43	
RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	8B	
RLB_SPI: SI		90	18	3	
RLB_SPI: SO	45	43	43	43	
RLB_SPI: CMDW	18	9	25	
RLB_SPI: SI				32	
RLB_SPI: SO	45	47	47	47	
RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	8B	
RLB_SPI: SI		90	18	3	
RLB_SPI: SO	45	43	43	43	
RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	18	9	44	
RLB_SPI: SI				14	24	
RLB_SPI: SO	45	47	47	47	47	
RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	8C	
RLB_SPI: SI		16	0	FF	20	40	0	0	
RLB_SPI: SO	45	43	43	43	43	43	43	43	
RLB_SPI: CMDW	8E	
RLB_SPI: SI		1C	80	
RLB_SPI: SO	45	43	43	
Radio        : +
PSRAM        : +
Display      : +
Sd Card      : -
Radio Initializing ... success!
RLB_SPI: CMDW	86	
RLB_SPI: SI		BE	CE	C4	
RLB_SPI: SO	45	43	43	43	
RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	8B	
RLB_SPI: SI		90	34	3	
RLB_SPI: SO	45	43	43	43	
RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	8B	
RLB_SPI: SI		C0	34	3	
RLB_SPI: SO	45	43	43	43	
RLB_SPI: CMDW	18	9	25	
RLB_SPI: SI				32	
RLB_SPI: SO	45	47	47	47	
RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	8B	
RLB_SPI: SI		C0	34	2	
RLB_SPI: SO	45	43	43	43	
RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	18	9	44	
RLB_SPI: SI				A4	B4	
RLB_SPI: SO	45	47	47	47	47	
RLB_SPI: CMDW	8E	
RLB_SPI: SI		15	80	
RLB_SPI: SO	45	43	43	
RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	8C	
RLB_SPI: SI		18	0	FF	20	40	0	0	
RLB_SPI: SO	45	43	43	43	43	43	43	43	
RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	8C	
RLB_SPI: SI		18	0	FF	0	40	0	0	
RLB_SPI: SO	45	43	43	43	43	43	43	43	
Radio Sending first packet ... RLB_SPI: CMDR	3	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	47	1	
RLB_SPI: CMDW	8C	
RLB_SPI: SI		18	0	1	0	40	0	0	
RLB_SPI: SO	45	43	43	43	43	43	43	43	
RLB_SPI: CMDW	8E	
RLB_SPI: SI		15	80	
RLB_SPI: SO	45	43	43	
RLB_SPI: CMDW	8F	
RLB_SPI: SI		0	0	
RLB_SPI: SO	45	43	43	
RLB_SPI: CMDW	1A	0	
RLB_SPI: SI			30	
RLB_SPI: SO	45	47	47	
RLB_SPI: CMDW	8D	
RLB_SPI: SI		40	1	0	1	0	0	0	0	
RLB_SPI: SO	45	43	43	43	43	43	43	43	43	
RLB_SPI: CMDW	97	
RLB_SPI: SI		FF	FF	
RLB_SPI: SO	45	43	43	
RLB_SPI: CMDW	83	
RLB_SPI: SI		0	0	0	
RLB_SPI: SO	45	43	43	43	

</p>
</details> 

 - MCU: ESP32S3 (Lilygo T3-S3)
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type: SX1280PA
 - Arduino IDE version: 2.3.4
 - Library version: 7.1.2

Nikita-Kononov, 2025-04-17:
I'm using the onboard USB-C to power both boards from my laptop. Could that be the issue? I will have access to a proper power supply in the next few days and I will try to power them from there.

jgromes, 2025-04-17:
> The code is too long because it's a universal code used for multiple boards that they produce.

I understand that, but because of this, neither of us understands this code. So I would recommend to use plain RadioLib examples, and learn with those.

> Could you elaborate on what the CMD_SET_TX command does?

It initiates the transmission, at which point the power consumption of the SX1280 will be the highest. So if there is a power issue, this is exactly where the MCU would get stuck, or reset.

There is a rather prominent warning on the [Lilygo site](https://lilygo.cc/products/t3s3-v1-0) that output power more than 3 dBm will damage the PA. RadioLib configures 10 dBm by default (if you do not set any arguments to `radio.begin()`), so you should be careful not to transmit with more than the 3 dBm output, and with a proper antenna connected, of course.

Also, since the same issue manifests in a different library, it seems very unlikely it is due to a software problem in RadioLib, so I will convert this into a discussion.