jgromes/RadioLib #1491: scanGuard wait delay to long

pchwalek, 2025-04-20:
I've been going through the LoRaWAN code and comparing my radio power consumption with a power analyzer to get more accurate timings when the radio is in TX/RX modes. I expect RX1 to be centered, with the overall size in us being the expected getTimeOnAir(0) + scanGuard*1000. Is this the intended architecture?  

If so, I think the wait length before starting RX1 should have the scanGuard divided by 2 subtracted from it instead of the [complete scanGuard value](https://github.com/jgromes/RadioLib/blob/c570b44dc35f3d2cddf326926b17b1ea91e32328/src/protocols/LoRaWAN/LoRaWAN.cpp#L1444).

Otherwise, the RX window timeout occurs before the packet is received, assuming there are slight time differences between your setup and the intended operation.

StevenCellist, 2025-04-21:
Before we go down a deep rabbit hole that this is: what do you mean by "the results are more consistent"? Have you had situations where you did not receive a downlinks that did actually occur, and did you improve the situation by that change?

The current implementation is fully intentional, which I can explain, but I would first like to clear up the above.

pchwalek, 2025-04-21:
Meaning that I am seeing the RX1 window centered at the 5s delay from the initial transmit window with my change applied. 

What I was seeing before was RADIOLIB_ERR_NO_JOIN_ACCEPT when attempting OTAA (i.e., US915, subband 2, LoRaWAN 1.1). I thought it could have been me being too close to the gateway, as per other threads, but when looking at the radio state as seen from the power analyzer, I was seeing the RX1 window shifted so that most of it occurred before the 5s threshold after the TX window. Looking at the code, it looks like how the scanguard is applied, it biases towards increasing the RX window earlier rather than equally in both time directions from the 5s threshold.

StevenCellist, 2025-04-21:
We haven't seen any reports of this behaviour before, even with my 50+ devices that I have deployed now. What hardware do you use? And do you use RadioLib 7.1.2, or upstream from GitHub? Because there has been a massive change in timing since 7.1.2 as we are now using staged modes instead of one big `startTransmit()` call.

pchwalek, 2025-04-22:
I am using 7.1.2 with an SX1262 radio on a custom board. The system is set up for US915, subband 2, and the radio settings are the below: 

`radio.XTAL = true;
 radio.standbyXOSC = true;
 state = radio.begin(915, 125, 12, 7, RADIOLIB_SX126X_SYNC_WORD_PRIVATE, 20, 8, 0, false)
 state = radio.setOutputPower(22);
 state = radio.setPaConfig(0x04, 0, 0x07, RADIOLIB_SX126X_PA_CONFIG_PA_LUT);
 state = radio.setPaRampTime(SX126X_RAMP_40_US);`

My mental model of the code for my configuration after the transmit window:
1. setup radio for RX1
2. calculate RX1 time by estimating RX time required to receive activation packet with an added scanGuard tolerance. 
3. delay 5 seconds minus the scanGuard and the time delta from the end of the transmission
4. put radio in RX mode for the length of time calculated in 2

From what I am seeing so far on the power analyzer, it looks like at the 5-second mark from the transmission, the approximate total value of the scanGuard is before the 5-second threshold, and what is after is the approximate estimate of the RX time required to receive the activation packet. This tracks with my current knowledge of the code where the full scanGuard is subtracted from the wait length. If subtracting half the scanGuard, the RX1 window will be centered around the 5 second threshold.

HeadBoffin, 2025-04-22:
> 7.1.2 with an SX1262 radio on a custom board

7.1.2 needs updating, the radio is not so important - **the MCU is everything here** - it is super important you tell us what it is so we can consider its timing abilities, see below.

As you are using a private sync word, knowing what LNS you are using, version and on what platform is important too.

> If so, I think the wait length before starting RX1 should have the scanGuard divided by 2 subtracted 
> from it instead of the complete scanGuard value

Please consider in your thinking that the stack works for many people, it's been throughly tested and has passed all the LW compliance tests. That's not to say that there aren't things to consider and improve. Many stacks, LMIC particularly on older AVR, had a heck of a job to get the scanGuard equivalent working - with lots of code for percentage adjustment.

Being late to listen is the main issue this covers - rarely would a gateway transmit so late that it is caught in the second half of the scanGuard.

As you have the expertise, the best efforts would be on sleeping between end of Tx and just before Rx1 as a common piece of functionality.

pchwalek, 2025-04-22:
I am using a Seeed Studio M2 gateway interfaced with TTN. The server has the device configured with the below:

- 902-928 MHz, FSB 2 (used by TTN)
- LoRaWAN Specification 1.1.0
- RP001 Regional Parameters 1.1 revision B

My understanding is that the scanGuard helps in diagnosing timing issues, more so on the device end. For example, if a user implemented the delay function in the HAL incorrectly (e.g., if their timer is misconfigured) or if there is an issue with the crystal/oscillator. Regarding delays in the reception of RX1, long distance and rain can increase ToA, which is why I think this is important.  

I understand that it is working for many people, and if this is a valid issue, it could be a point of improvement for the future. =)

pchwalek, 2025-04-22:
O, sorry. STM32U595 using HSI, and for testing, I am using a scanGuard of 100ms for easy viewing of RX1 with an analyzer.

StevenCellist, 2025-04-22:
The HSI is, as far as I've seen, never a good choice for LW on STM32. As far as the datasheet is concerned, the HSI at 16MHz has a .5% accuracy, with the HSI at 48MHz even up to 2%. So that will be your main problem to resolve.
The `scanGuard` is part of a solution which indeed would benefit from moving over a bit like you suggested - I will change that - but I highly recommend running off the RTC or LSE which is much more time-accurate. And as you found out, LW is highly sensitive to clock accuracy.

On a side note: do you use RadioLib through a dedicated HAL, or through STM32duino?

HeadBoffin, 2025-04-22:
> The server has the device configured with the below:

Which server - rather like the MCU, pertinent info - **so what is the LNS**?

> My understanding is that the scanGuard helps in diagnosing timing issues,

Not really. More a mechanism to allow the use of lower cost MCU's that don't have a rock solid time source.

> if a user implemented the delay function in the HAL incorrectly (e.g., if their timer is misconfigured)

These users are very very rare - and mostly people use the supplied HAL tick from the vendors. Again, perhaps your experience is different from the typical user base.

> Regarding delays in the reception of RX1, long distance and rain can increase ToA, which is why I think this is important.

For the former, sure, there's 33ms difference between 100m and 10,000m, which the scanGuard could compensate for, but there would be a much longer preamble on a the longer range which would use a higher spreading factor - think 40ms vs almost 1000ms, so even if the Rx window opens mildly late, on longer ranges this is less of an issue.

Whilst I have seen dramatic attenuation of signal strength due to heavy rain, I've not seen it alter the speed of light. Where do you live?

pchwalek, 2025-04-22:
Thanks, @StevenCellist! Noted on the clock inaccuracies. I'm eventually switching to an HSE crystal, but its been omitted in this electrical design, but I can try RTC in the short term. 

I'm running barebone and created my own HAL. 

@HeadBoffin 

> Which server - rather like the MCU, pertinent info - so what is the LNS?

Maybe I'm not understanding the question but the LNS I am using is TTN, specifically nam1.cloud.thethings.network v3.34.1.2363763fbe.

> Not really. More a mechanism to allow the use of lower cost MCU's that don't have a rock solid time source.

Right, so for users using cheaper MCUs, such as clones, its not clear if their timing is running faster or slower than the gateway clock. 

> These users are very very rare - and mostly people use the supplied HAL tick from the vendors. Again, perhaps your experience is different from the typical user base.

I am using the HAL_GetTick() STM32 API for the delay function, and the clock setup is relatively standard.

> For the former, sure, there's 33ms difference between 100m and 10,000m, which the scanGuard could compensate for, but there would be a much longer preamble on a the longer range which would use a [...]

> Whilst I have seen dramatic attenuation of signal strength due to heavy rain, I've not seen it alter the speed of light. Where do you live?

I live in the Northeast U.S., and what I meant by rain, as well as any other medium (e.g., trees), between an RF source and receiver, is that this can introduce scattering effects where a reflection could be the signal that is received and decoded. You're right, the speed of sound doesn't change.

HeadBoffin, 2025-04-22:
> but the LNS I am using is TTN

Which uses TTS, an LNS regularly updated with the aspiration of being the most compliant. So it will instruct the gateway to deliver the downlink at the right time - because it does that on my TTN/TTI gateways. And with a mix of vendor brands, including some of my own build, they just work.

> Right, so for users using cheaper MCUs, such as clones, its not clear if their timing is running faster or slower than the gateway clock.

No. Cheaper MCU's are not necessarily clones, and even if they all were, they just may not be able to keep an accurate clock or have some temperature drift issues, or the designer may choose totally the wrong clock to use or not include a crystal or, even better, a TCXO. LW was been around for a while, using an Arduino ProMini at 8MHz still works fine at the expense of having a a wider scan guard.

> is that this can introduce scattering effects where a reflection could be the signal that is received

Rain not so much, large flat buildings in urban areas, sure.

> You're right, the speed of sound doesn't change.

Um, it does, with both temperature & air density (altitude), which has something of a correlation but there are some temperature zones as well as inversion layers in the atmosphere. But we don't use sound waves for LoRa so that doesn't apply.

It seems you have a working solution to a somewhat self-generated issue. And at some point @StevenCellist will refine some code in the context of his detailed knowledge of the LW spec & his code base & the potential costs of re-running the certification tests and pass it over to some poor sap to test. But for now, this issue appears closable ...

pchwalek, 2025-04-22:
I'm not interested in defending semantics, but I don't believe it is a self-generated issue. I'll go ahead and close out this comment since it's been acknowledged. Thank you!

StevenCellist, 2025-05-01:
+1, this problem was hit during end-device certification test where a payload of maximum size is sent but dropped because the window closed too early. Will also ever so slightly increase the scanGuard value.

StevenCellist, 2025-05-06:
After some proper digging around as well as testing FSK (which uses bizarrely small Rx windows), this issue is fixed in the commit referenced above. A new variable is introduced which tracks the duration of the SPI transaction that launches a certain mode in the radio (which still takes a few milliseconds), allowing more precise timing of the window opening. This launch duration was the main reason why the window was heavily biased to opening 'too soon', but as we now know how long this launch takes, we can properly center around the actual real Rx window.

closed 2025-05-06 by StevenCellist

pchwalek, 2025-05-08:
@StevenCellist, happy to share it, but there are a few things that are specific to my application in the HAL. =)

[STM32Hal.cpp.zip](https://github.com/user-attachments/files/20095941/STM32Hal.cpp.zip)