jgromes/RadioLib #1496: SX127x: FSK timeout interrupt

StevenCellist, 2025-05-01:
While testing LoRaWAN with FSK, I noticed that the RxSingle functionality appears to be missing from FSK. I see the following comment in its driver:

https://github.com/jgromes/RadioLib/blob/c570b44dc35f3d2cddf326926b17b1ea91e32328/src/modules/SX127x/SX127x.cpp#L1700-L1701

However, I do not think this is right, since in the [datasheet](https://cdn-shop.adafruit.com/product-files/3179/sx1276_77_78_79.pdf) sec. 4.2.3.9 / page 54, I see the following: 

[image]

As such, I do think that it is possible to set a Timeout value and use it to generate a Timeout interrupt, especially in absence of `PreambleDetect`. Would it be possible to investigate this?

(Note: I don't have an SX127x myself at this moment. The tests were performed by another gentleman. )

StevenCellist, 2025-05-01:
Steps to verify the problem/solution for LoRaWAN on SX127x (likely):
- Modify one of the default channels, such as https://github.com/jgromes/RadioLib/blob/c570b44dc35f3d2cddf326926b17b1ea91e32328/src/protocols/LoRaWAN/LoRaWANBands.cpp#L35  
to use `drMax = 7`.
- After a successful join, configure `node.setDatarate(7)` to move to FSK datarate for uplink and Rx1 window.
- Observe protocol debug output. Instead of showing `Downlink missing!` (current behaviour) after closing Rx1, it should open Rx2 (and switch back and forth to LoRa modem).

Note that the above communication will not result in any valid uplinks, as it is not a valid FSK channel. That shouldn't matter however, as we wouldn't want to receive a downlink anyway to yield an RxTimeout instead.

jgromes, 2025-05-03:
@StevenCellist I did some more digging, and unfortunately it looks like this won't be as easy as I originally thought. SX127x interrupt configuration is not as flexible as on SX126x, timeout can only be signalled on DIO2 or DIO4 which are left unconnected on all SX127x off-the-shelf modules I have seen (that's why the RadioLib `Module` interface only supports DIO0 and DIO1). 

So unfortunately I don't thin this is possible without some sort of timer or loop running on the host MCU.

jgromes, 2025-05-03:
On further thinking, this might *just about* work for LoRaWAN - it always checks the timeout interrupt:

https://github.com/jgromes/RadioLib/blob/c570b44dc35f3d2cddf326926b17b1ea91e32328/src/protocols/LoRaWAN/LoRaWAN.cpp#L1458-L1459

We can use the SX127x timeout mechanism even without having access to the physical DIO pin. Quick check with plain FSK reception withouth LoRaWAN shows it is working:

```
19:28:08.311 -> [SX1278] Received packet!
19:28:08.357 -> [SX1278] Data:		Hello Word! #361
19:28:08.357 -> [SX1278] RSSI:		-78.50 dBm
19:28:08.404 -> [SX1278] SNR:		0.00 dB
19:28:08.450 -> [SX1278] Frequency error:	0.00 Hz
19:28:08.497 -> IRQ = 0x0000 (0)
19:28:08.497 -> IRQ = 0x0000 (0)
19:28:08.497 -> IRQ = 0x0000 (0)
19:28:08.497 -> IRQ = 0x0000 (0)
19:28:08.543 -> IRQ = 0x0000 (0)
19:28:08.590 -> IRQ = 0x0000 (0)
19:28:08.590 -> IRQ = 0x0000 (0)
19:28:08.590 -> IRQ = 0x0000 (0)
19:28:08.590 -> IRQ = 0x0000 (0)
19:28:08.636 -> IRQ = 0x0000 (0)
19:28:08.636 -> IRQ = 0x0000 (0)
19:28:08.682 -> IRQ = 0x0000 (0)
19:28:08.682 -> IRQ = 0x0000 (0)
19:28:08.729 -> IRQ = 0x0004 (4)
19:28:08.729 -> IRQ = 0x0004 (4)
19:28:08.729 -> IRQ = 0x0004 (4)
```

The log shows that approximately 230 ms after restarting Rx post packet reception, the timeout flag in `RADIOLIB_SX127X_REG_IRQ_FLAGS_1` is indeed raised.

StevenCellist, 2025-05-03:
Living on the (rising) edge 😉 but would be sweet if it indeed works!

jgromes, 2025-05-04:
@StevenCellist partial success, at least I think. I tested using the approach you proposed (by chaging one of the EU868 channels to support FSK data rate). In the first log, Rx timeout is not emulated and indeed there's that `Downlink missing!` message. In the second log, there's an Rx timeout and the radio then tries to switch to LoRa. However, for some reason it fails to do so and returns -16, so I'm guessing there's some transition to standby missing.

<details><summary>Without timeout</summary>
<p>

```
16:30:27.891 -> [SX1278] Initializing ... [3.125609] RLB_DBG: 
16:30:27.891 -> RadioLib Info
16:30:27.891 -> Version:  "7.1.2.0"
16:30:27.891 -> Platform: "ESP32"
16:30:27.891 -> "May  4 2025" "15:38:26"
16:30:27.891 -> [3.132173] RLB_DBG: M	SX127x
16:30:27.891 -> success!
16:30:27.891 -> [LoRaWAN] Attempting over-the-air activation ... success!
16:30:27.891 -> [LoRaWAN] Joining the network ... [3.146394] RLB_PRO: Setting up dynamic channels
16:30:27.891 -> [3.146488] RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
16:30:27.891 -> [3.157175] RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
16:30:27.938 -> [3.157412] RLB_PRO: UL:   2 1 868.500 (0 - 7) | DL:   2 1 868.500 (0 - 7)
16:30:27.938 -> [3.167798] RLB_PRO: [MAC] 0x03
16:30:27.938 -> [3.167963] RLB_PRO: 00000000: 30                                               0                 
16:30:27.938 -> [3.178227] RLB_PRO: LinkAdrReq: dataRate = 3, txSteps = 0, nbTrans = 0
16:30:27.938 -> [3.178632] RLB_PRO: LinkAdrAns: 07
16:30:27.938 -> [3.188846] RLB_PRO: [MAC] 0x04
16:30:27.938 -> [3.188934] RLB_PRO: 00000000: 07                                               .                 
16:30:27.938 -> [3.199181] RLB_PRO: DutyCycleReq: max duty cycle = 1/2^7
16:30:27.938 -> [3.199251] RLB_PRO: [MAC] 0x05
16:30:27.984 -> [3.209459] RLB_PRO: 00000000: 00 d2 ad 84                                      ....              
16:30:27.984 -> [3.209571] RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525
16:30:27.984 -> [3.220939] RLB_PRO: [MAC] 0x08
16:30:27.984 -> [3.221048] RLB_PRO: 00000000: 01                                               .                 
16:30:27.984 -> [3.231273] RLB_PRO: RXTimingSetupReq: delay = 1 sec
16:30:27.984 -> [3.241520] RLB_PRO: [MAC] 0x09
16:30:27.984 -> [3.241594] RLB_PRO: 00000000: 05                                               .                 
16:30:27.984 -> [3.251888] RLB_PRO: [MAC] 0x0c
16:30:27.984 -> [3.251951] RLB_PRO: 00000000: 65                                               e                 
16:30:28.031 -> [3.262232] RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
16:30:28.031 -> [3.262337] RLB_PRO: [MAC] 0x0f
16:30:28.031 -> [3.272571] RLB_PRO: 00000000: fa                                               .                 
16:30:28.031 -> [3.272678] RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
16:30:28.077 -> [3.324934] RLB_PRO: 
16:30:28.077 -> [3.325017] RLB_PRO: PHY:  Frequency = 868.100 MHz, TX = 16 dBm
16:30:28.077 -> [3.327177] RLB_PRO: LoRa: SF = 9, BW = 125.0 kHz, CR = 4/5, IQ: U
16:30:28.310 -> [3.536084] RLB_PRO: JoinRequest sent (DevNonce = 0) <-- Rx Delay start
16:30:28.310 -> [3.536364] RLB_PRO: 00000000: 00 ee ff c0 b0 11 10 ad 12 25 e1 05 d0 7e d5 b3  .........%%...~..
16:30:28.310 -> [3.546725] RLB_PRO: 00000010: 70 00 00 49 3a 82 59                             p..I:.Y           
16:30:28.310 -> [3.558103] RLB_PRO: 
16:30:28.310 -> [3.558190] RLB_PRO: PHY:  Frequency = 868.100 MHz, TX = 16 dBm
16:30:28.310 -> [3.560088] RLB_PRO: LoRa: SF = 9, BW = 125.0 kHz, CR = 4/5, IQ: D
16:30:33.290 -> [8.531274] RLB_PRO: Opening Rx1 window (111 ms timeout)... <-- Rx Delay end 
16:30:33.383 -> [8.642048] RLB_PRO: Closing Rx1 window
16:30:33.523 -> [8.785942] RLB_PRO: JoinAccept (JoinNonce = 12, previously 0):
16:30:33.523 -> [8.786213] RLB_PRO: 00000000: 20 0c 00 00 13 00 00 45 1e 0b 26 80 05 18 4f 84   ......E..&...O.
16:30:33.569 -> [8.796619] RLB_PRO: 00000010: e8 56 84 b8 5e 84 88 66 84 58 6e 84 00 63 c0 fa  .V..^..f.Xn..c..
16:30:33.569 -> [8.806867] RLB_PRO: 00000020: 07                                               .                 
16:30:33.569 -> [8.806970] RLB_PRO: LoRaWAN revision: 1.1
16:30:33.569 -> [8.818774] RLB_PRO: Setting up dynamic channels
16:30:33.569 -> [8.818834] RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
16:30:33.569 -> [8.829359] RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
16:30:33.569 -> [8.829590] RLB_PRO: UL:   2 1 868.500 (0 - 7) | DL:   2 1 868.500 (0 - 7)
16:30:33.616 -> [8.839991] RLB_PRO: [MAC] 0x05
16:30:33.616 -> [8.840109] RLB_PRO: 00000000: 00 d2 ad 84                                      ....              
16:30:33.616 -> [8.850330] RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525
16:30:33.616 -> [8.861562] RLB_PRO: [MAC] 0x08
16:30:33.616 -> [8.861672] RLB_PRO: 00000000: 05                                               .                 
16:30:33.616 -> [8.871884] RLB_PRO: RXTimingSetupReq: delay = 5 sec
16:30:33.616 -> [8.871950] RLB_PRO: Processing CFList
16:30:33.616 -> [8.872021] RLB_PRO: [MAC] 0x07
16:30:33.616 -> [8.882250] RLB_PRO: 00000000: 03 18 4f 84 50                                   ..O.P             
16:30:33.662 -> [8.893432] RLB_PRO: UL:   3 1 867.100 (0 - 5) | DL:   3 1 867.100 (0 - 5)
16:30:33.662 -> [8.893734] RLB_PRO: [MAC] 0x07
[... 146 lines cut ...]
16:26:19.416 -> [8.813689] RLB_PRO: 
[... cut ...]
16:26:54.507 -> [43.870365] RLB_PRO: 00000000: 0b 01                                            ..                
16:26:54.554 -> [43.921583] RLB_PRO: Uplink (FCntUp = 1) decoded:
16:26:54.554 -> [43.921779] RLB_PRO: 00000000: 49 00 00 00 00 00 25 cc 0b 26 00 00 00 00 00 1a  I.....%%..&......
16:26:54.554 -> [43.932125] RLB_PRO: 00000010: 40 25 cc 0b 26 82 01 00 66 51 0a bf c7 0a 0b 46  @%%..&...fQ.....F
16:26:54.554 -> [43.932332] RLB_PRO: 00000020: 47 d6 79 ee 16 e6 d7 8e 5d e1 64 2b b3 74        G.y.....].d+.t    
16:26:54.554 -> failed, code -1108
</p>
</details>

StevenCellist, 2025-05-04:
Hmm, that's a bit weird:

https://github.com/jgromes/RadioLib/blob/c570b44dc35f3d2cddf326926b17b1ea91e32328/src/protocols/LoRaWAN/LoRaWAN.cpp#L2857

If need be, I can look for an SX127x module to try it myself.

jgromes, 2025-05-04:
Managed to track down the -16 issue. When switching the modem from FSK to LoRa, we need to write into the opmode register (address 0x01). Bit 7 controls the modem (0 for FSK, 1 for LoRa). However, from previous usage, it looks like bit 3 (low frequency mode access) remains set, despite the device being reset via the reset pin and being in sleep mode (requirement for the modem change). Bit weird, but after switching the modem that bit goes to 0 which confuses RadioLib SPI check since the value in the register changes without our input. The fix is to not check this bit when switching modem.

```
18:41:44.361 -> [15.862553] RLB_SPI: R	1	8	
18:41:44.361 -> [15.862674] RLB_SPI: W	1	88	
18:41:44.361 -> [15.862797] RLB_SPI: R	1	80	
18:41:44.361 -> [15.873073] RLB_SPI: 
18:41:44.361 -> [15.873113] RLB_SPI: address:	0x1
18:41:44.361 -> [15.873154] RLB_SPI: bits:		7 7
18:41:44.361 -> [15.873211] RLB_SPI: value:		0x80
18:41:44.361 -> [15.883391] RLB_SPI: current:	0x8
18:41:44.361 -> [15.883435] RLB_SPI: mask:		0x80
18:41:44.407 -> [15.883474] RLB_SPI: new:		0x88
18:41:44.407 -> [15.883515] RLB_SPI: read:		0x80
``` 

Now it seems that the modem gets switched correctly. You can see the first Rx window is FSK, the second one is back to LoRa. However, this only happens in the first uplink/downlink loop, from second loop onwards it only opens the first FSK window, which I'm guessing is not the desired behavior - see below.

<details><summary>Log</summary>
<p>

```
18:56:46.953 -> [SX1278] Initializing ... [1.324155] RLB_DBG: 
18:56:46.953 -> RadioLib Info
18:56:46.953 -> Version:  "7.1.2.0"
18:56:46.953 -> Platform: "ESP32"
18:56:46.953 -> "May  4 2025" "18:53:21"
18:56:46.953 -> [1.330175] RLB_DBG: M	SX127x
18:56:46.953 -> success!
18:56:46.999 -> [LoRaWAN] Attempting over-the-air activation ... success!
18:56:46.999 -> [LoRaWAN] Joining the network ... [1.344851] RLB_PRO: Setting up dynamic channels
18:56:46.999 -> [1.344942] RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
18:56:46.999 -> [1.355631] RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
18:56:46.999 -> [1.355872] RLB_PRO: UL:   2 1 868.500 (0 - 7) | DL:   2 1 868.500 (0 - 7)
18:56:46.999 -> [1.366254] RLB_PRO: [MAC] 0x03
18:56:46.999 -> [1.366419] RLB_PRO: 00000000: 30                                               0                 
18:56:46.999 -> [1.376683] RLB_PRO: LinkAdrReq: dataRate = 3, txSteps = 0, nbTrans = 0
18:56:47.046 -> [1.377088] RLB_PRO: LinkAdrAns: 07
18:56:47.046 -> [1.387302] RLB_PRO: [MAC] 0x04
18:56:47.046 -> [1.387394] RLB_PRO: 00000000: 07                                               .                 
18:56:47.046 -> [1.397638] RLB_PRO: DutyCycleReq: max duty cycle = 1/2^7
18:56:47.046 -> [1.397709] RLB_PRO: [MAC] 0x05
18:56:47.046 -> [1.407916] RLB_PRO: 00000000: 00 d2 ad 84                                      ....              
18:56:47.046 -> [1.408028] RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525
18:56:47.046 -> [1.419392] RLB_PRO: [MAC] 0x08
18:56:47.046 -> [1.419501] RLB_PRO: 00000000: 01                                               .                 
18:56:47.092 -> [1.429729] RLB_PRO: RXTimingSetupReq: delay = 1 sec
18:56:47.092 -> [1.439981] RLB_PRO: [MAC] 0x09
18:56:47.092 -> [1.440061] RLB_PRO: 00000000: 05                                               .                 
18:56:47.092 -> [1.450343] RLB_PRO: [MAC] 0x0c
18:56:47.092 -> [1.450413] RLB_PRO: 00000000: 65                                               e                 
18:56:47.092 -> [1.460688] RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
18:56:47.092 -> [1.460797] RLB_PRO: [MAC] 0x0f
18:56:47.092 -> [1.471028] RLB_PRO: 00000000: fa                                               .                 
18:56:47.139 -> [1.471139] RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
18:56:47.185 -> [1.522945] RLB_PRO: 
18:56:47.185 -> [1.523032] RLB_PRO: PHY:  Frequency = 868.500 MHz, TX = 16 dBm
18:56:47.185 -> [1.525194] RLB_PRO: LoRa: SF = 9, BW = 125.0 kHz, CR = 4/5, IQ: U
18:56:47.372 -> [1.734203] RLB_PRO: JoinRequest sent (DevNonce = 0) <-- Rx Delay start
18:56:47.372 -> [1.734487] RLB_PRO: 00000000: 00 ee ff c0 b0 11 10 ad 12 25 e1 05 d0 7e d5 b3  .........%%...~..
18:56:47.372 -> [1.744849] RLB_PRO: 00000010: 70 00 00 49 3a 82 59                             p..I:.Y           
18:56:47.418 -> [1.756243] RLB_PRO: 
18:56:47.418 -> [1.756346] RLB_PRO: PHY:  Frequency = 868.500 MHz, TX = 16 dBm
18:56:47.418 -> [1.758247] RLB_PRO: LoRa: SF = 9, BW = 125.0 kHz, CR = 4/5, IQ: D
18:56:52.392 -> [6.729277] RLB_PRO: Opening Rx1 window (111 ms timeout)... <-- Rx Delay end 
18:56:52.485 -> [6.840052] RLB_PRO: Closing Rx1 window
18:56:52.625 -> [6.984063] RLB_PRO: JoinAccept (JoinNonce = 18, previously 0):
18:56:52.625 -> [6.984330] RLB_PRO: 00000000: 20 12 00 00 13 00 00 62 ba 0b 26 80 05 18 4f 84   ......b..&...O.
18:56:52.625 -> [6.994737] RLB_PRO: 00000010: e8 56 84 b8 5e 84 88 66 84 58 6e 84 00 a8 6f a8  .V..^..f.Xn...o.
18:56:52.625 -> [7.004981] RLB_PRO: 00000020: 0a                                               .                 
[... 68 lines cut ...]
18:57:29.683 -> [44.019585] RLB_PRO: Uplink (FCntUp = 1) decoded:
18:57:29.683 -> [44.019786] RLB_PRO: 00000000: 49 00 00 00 00 00 62 ba 0b 26 00 00 00 00 00 1a  I.....b..&......
18:57:29.683 -> [44.030128] RLB_PRO: 00000010: 40 62 ba 0b 26 82 01 00 3c cf 0a e6 54 7b 99 71  @b..&...<...T{.q
[... cut ...]
18:57:29.730 -> [44.077478] RLB_PRO: Uplink sent <-- Rx Delay start
18:57:29.730 -> [44.078135] RLB_PRO: 
18:57:29.730 -> [44.078191] RLB_PRO: PHY:  Frequency = 868.500 MHz, TX = 16 dBm
18:57:29.730 -> [44.089905] RLB_PRO: FSK:  BR = 50.0, FD = 25.0 kHz
18:57:34.707 -> [49.072269] RLB_PRO: Opening Rx1 window (8 ms timeout)... <-- Rx Delay end 
18:57:34.707 -> [49.080052] RLB_PRO: Closing Rx1 window
18:57:34.753 -> [49.120006] RLB_PRO: Downlink missing!
18:57:34.753 -> no downlink!
``` 

</p>
</details>

StevenCellist, 2025-05-04:
Progress, but not an easy journey.. indeed the expected behaviour should be for the node to go on with up/Rx1/Rx2 without hiccups.
Cheering you on from the sidelines! 😉

jgromes, 2025-05-04:
Must be working since I realized the problem was that `SX127x::calculateRxTimeout` only handled LoRa, not FSK. After fixing that, it now correctly toggles between Rx1/Rx2. Notice the `timeout` and `timeoutUs` prints, which the values calculated in `SX127x::calculateRxTimeout`.

@StevenCellist is there a way to convince a gateway/TTN to use FSK so that I can see actual downlinks? I'd like to check that before pushing this.

<details><summary>Details</summary>
<p>

```
20:34:59.752 -> [SX1278] Initializing ... [6.577675] RLB_DBG: 
20:34:59.752 -> RadioLib Info
20:34:59.752 -> Version:  "7.1.2.0"
20:34:59.752 -> Platform: "ESP32"
20:34:59.752 -> "May  4 2025" "18:53:21"
20:34:59.752 -> [6.584123] RLB_DBG: M	SX127x
20:34:59.752 -> success!
20:34:59.752 -> [LoRaWAN] Attempting over-the-air activation ... success!
20:34:59.752 -> [LoRaWAN] Joining the network ... [6.598365] RLB_PRO: Setting up dynamic channels
20:34:59.752 -> [6.598469] RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
20:34:59.798 -> [6.609136] RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
20:34:59.798 -> [6.609359] RLB_PRO: UL:   2 1 868.500 (0 - 7) | DL:   2 1 868.500 (0 - 7)
20:34:59.798 -> [6.619760] RLB_PRO: [MAC] 0x03
20:34:59.798 -> [6.619931] RLB_PRO: 00000000: 30                                               0                 
20:34:59.798 -> [6.630199] RLB_PRO: LinkAdrReq: dataRate = 3, txSteps = 0, nbTrans = 0
20:34:59.798 -> [6.630593] RLB_PRO: LinkAdrAns: 07
20:34:59.798 -> [6.640808] RLB_PRO: [MAC] 0x04
20:34:59.798 -> [6.640902] RLB_PRO: 00000000: 07                                               .                 
20:34:59.798 -> [6.651148] RLB_PRO: DutyCycleReq: max duty cycle = 1/2^7
20:34:59.845 -> [6.651222] RLB_PRO: [MAC] 0x05
20:34:59.845 -> [6.661433] RLB_PRO: 00000000: 00 d2 ad 84                                      ....              
20:34:59.845 -> [6.661555] RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525
20:34:59.845 -> [6.672902] RLB_PRO: [MAC] 0x08
20:34:59.845 -> [6.673013] RLB_PRO: 00000000: 01                                               .                 
20:34:59.845 -> [6.683240] RLB_PRO: RXTimingSetupReq: delay = 1 sec
20:34:59.845 -> [6.693489] RLB_PRO: [MAC] 0x09
20:34:59.845 -> [6.693567] RLB_PRO: 00000000: 05                                               .                 
20:34:59.891 -> [6.703860] RLB_PRO: [MAC] 0x0c
20:34:59.891 -> [6.703930] RLB_PRO: 00000000: 65                                               e                 
20:34:59.891 -> [6.714200] RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
20:34:59.891 -> [6.714309] RLB_PRO: [MAC] 0x0f
20:34:59.891 -> [6.724546] RLB_PRO: 00000000: fa                                               .                 
20:34:59.891 -> [6.724659] RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
20:34:59.938 -> [6.776889] RLB_PRO: 
20:34:59.938 -> [6.776974] RLB_PRO: PHY:  Frequency = 868.500 MHz, TX = 16 dBm
20:34:59.938 -> [6.779105] RLB_PRO: LoRa: SF = 9, BW = 125.0 kHz, CR = 4/5, IQ: U
20:35:00.171 -> [6.988050] RLB_PRO: JoinRequest sent (DevNonce = 0) <-- Rx Delay start
20:35:00.171 -> [6.988326] RLB_PRO: 00000000: 00 ee ff c0 b0 11 10 ad 12 25 e1 05 d0 7e d5 b3  .........%%...~..
20:35:00.171 -> [6.998670] RLB_PRO: 00000010: 70 00 00 49 3a 82 59                             p..I:.Y           
20:35:00.171 -> [7.010070] RLB_PRO: 
20:35:00.171 -> [7.010178] RLB_PRO: PHY:  Frequency = 868.500 MHz, TX = 16 dBm
20:35:00.171 -> [7.012030] RLB_PRO: LoRa: SF = 9, BW = 125.0 kHz, CR = 4/5, IQ: D
20:35:00.171 -> [7.023080] RLB_PRO: timeout = 26
20:35:00.171 -> [7.023137] RLB_PRO: timeoutUs = 109000
20:35:05.150 -> [11.983233] RLB_PRO: Opening Rx1 window (111 ms timeout)... <-- Rx Delay end 
20:35:05.243 -> [12.094004] RLB_PRO: Closing Rx1 window
20:35:05.243 -> [12.094159] RLB_PRO: timedOut = 0
20:35:05.383 -> [12.237822] RLB_PRO: JoinAccept (JoinNonce = 22, previously 0):
20:35:05.430 -> [12.238099] RLB_PRO: 00000000: 20 16 00 00 13 00 00 3f f8 0b 26 80 05 18 4f 84   ......?..&...O.
20:35:05.430 -> [12.248500] RLB_PRO: 00000010: e8 56 84 b8 5e 84 88 66 84 58 6e 84 00 dd b1 9a  .V..^..f.Xn.....
20:35:05.430 -> [12.258743] RLB_PRO: 00000020: 85                                               .                 
20:35:05.430 -> [12.258857] RLB_PRO: LoRaWAN revision: 1.1
20:35:05.430 -> [12.270651] RLB_PRO: Setting up dynamic channels
20:35:05.430 -> [12.270717] RLB_PRO: UL:   0 1 868.100 (0 - 5) | DL:   0 1 868.100 (0 - 5)
20:35:05.430 -> [12.281144] RLB_PRO: UL:   1 1 868.300 (0 - 5) | DL:   1 1 868.300 (0 - 5)
20:35:05.476 -> [12.281362] RLB_PRO: UL:   2 1 868.500 (0 - 7) | DL:   2 1 868.500 (0 - 7)
20:35:05.476 -> [12.291775] RLB_PRO: [MAC] 0x05
20:35:05.476 -> [12.291894] RLB_PRO: 00000000: 00 d2 ad 84                                      ....              
20:35:05.476 -> [12.302116] RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq = 869.525
20:35:05.476 -> [12.313351] RLB_PRO: [MAC] 0x08
20:35:05.476 -> [12.313469] RLB_PRO: 00000000: 05                                               .                 
20:35:05.476 -> [12.323764] RLB_PRO: RXTimingSetupReq: delay = 5 sec
[... 118 lines cut ...]
20:36:19.288 -> [86.116691] RLB_PRO: FSK:  BR = 50.0, FD = 25.0 kHz
20:36:19.288 -> [86.117521] RLB_PRO: timeout = 21
20:36:19.288 -> [86.117576] RLB_PRO: timeoutUs = 6760
20:36:24.268 -> [91.099227] RLB_PRO: Opening Rx1 window (8 ms timeout)... <-- Rx Delay end 
[... cut ...]
20:36:24.315 -> [91.128481] RLB_PRO: PHY:  Frequency = 869.525 MHz, TX = 16 dBm
20:36:24.315 -> [91.130808] RLB_PRO: LoRa: SF = 12, BW = 125.0 kHz, CR = 4/5, IQ: D
20:36:24.315 -> [91.131831] RLB_PRO: timeout = 20
20:36:24.315 -> [91.142085] RLB_PRO: timeoutUs = 669000
20:36:25.292 -> [92.099233] RLB_PRO: Opening Rx2 window (671 ms timeout)... <-- Rx Delay end 
20:36:25.943 -> [92.770004] RLB_PRO: Closing Rx2 window
20:36:25.943 -> [92.770155] RLB_PRO: timedOut = 128
20:36:25.943 -> no downlink!
``` 

</p>
</details>

StevenCellist, 2025-05-04:
Nope, no chance for TTN. It will only receive FSK uplinks, not schedule them for downlinks.

I don't know for sure whether Chirpstack is able to use a normal gateway for FSK downlinks. I haven't seen any evidence of FSK downlinks myself so far. But now that we have confirmed that RadioLib is able to receive FSK downlinks I have more confidence to test it again. I can give you access to a self-hosted CS instance if you'd like to play with it yourself :)

StevenCellist, 2025-05-06:
I've verified that TTS still does not do FSK: gateway receives, TTS forwards to PacketBroker, but TTS doesn't process the FSK uplinks itself.
I just confirmed that Chirpstack on the other hand receives FSK uplinks and sends FSK downlinks: my SX1262-based device happily works on the FSK channel (after reworking the timing issues, so use of upstream is recommended).

closed 2025-05-28 by jgromes