jgromes/RadioLib #1524: [LoRaWAN] Bugs from LoRaWAN Certification Test Tool

StevenCellist, 2025-06-14:
With access to the newest version of the compliance test tool, some new items surface for the LoRaWAN stack.
 
* With the recent removal of `MaxFCntGap` from RadioLib (which is not defined in the specification), a replay attack will always result in an FCnt rollover. There is no deterministic way to prevent this, but ignoring FCntDown16 values that are slightly lower than the current FCntDown16 value (which is the most likely to happen during replay attack) will be a decent solution.
* Do not allow a value for `macTxSteps` that is greater than the `powerNumSteps` field in the band (i.e. reject RFU values). This was simply not implemented
* At the end of ADR backoff, the device should enable all default channels _on top of the already available channels_. The current situation is that all other channels are cleared.
* Still increment FCnt even after a Tx timeout, because it is rather likely that the transmission still occurred.
* LCTT really wants network uplinks (i.e. frames with MAC command and without user payload) to use the FRMPayload instead of piggybacking the MAC payload into FOpts. This potentially saves the FPort byte. The current situation is that MAC commands are always piggybacked into an application uplink as users are not allowed to send network uplinks in any other situation than to request LinkCheck or DeviceTime. Whether this is worth the implementation effort is debatable.

HeadBoffin, 2025-06-14:
The Venn diagram of Specification, Test Suite and Real Life is always illuminating & inevitably has some elements of compromise.

Fully documented deviations from the first two that could result in failure at a test house would preferable than any larger changes (like the last item) that could compromise what is proving to be a reliable stack.

StevenCellist, 2025-06-14:
* The grace period before triggering TxTimeout is too short for FSK uplinks, often resulting in false errors.

StevenCellist, 2025-06-16:
All listed issues were fixable without implementing any new logic: I could re-use existing structures.

Self-certified, passing all tests for Class A on dynamic channelplan.

Major milestone!! 🥳 

[image]

jgromes, 2025-06-16:
Amazing job @StevenCellist - indeed, this is a major accomplishment that you should be rightfully proud of!