jgromes/RadioLib #1534: ESP32-C6 MCU Restarting after Initializing RadioLib

adiiot-1316, 2025-06-22:
Hi Team, 

I am using Ebyte E28-2G4M27SX LORA module which is based on SX-1281 (https://www.cdebyte.com/products/E28-2G4M27SX) and interfacing it with Waveshare ESP32-C6-Devkit-N8(https://www.waveshare.com/esp32-c6-dev-kit-n8.htm). 
Issue is that as soon as code `radio.begin()` is executed, it is returning state `RADIOLIB_ERR_NONE` and post that MCU restarts. 

## Ebyte E28-2G4M27SX Pinout

[image]

## Wiring Diagram

| EBYTE Pin | ESP32-C6 Pin|
| -------- | ------- |
| Vcc | 3.3V |
| Gnd | Gnd |
| MISO Tx | GPIO20 |
| MOSI Rx | GPIO19 |
|  SCK_RTSN | GPIO21 |
| NSS_CTS | GPIO 18 |
| GND | - |
| RX | GPIO22 |
| GND |  - |
| DIO3 | - |
|  DIO2 | - |
| DIO1 | GPIO11 |
| BUSY | GPIO10 |
| NRESET | GPIO3 |
| GND | - |
| TX | GPIO23 |

## Code
```
#include <RadioLib.h>
#include <SPI.h>
SX1281 radio = new Module(18, 11, 3, 10);
void setup() {
  Serial.begin(115200);
  SPI.begin(21, 20, 19, 18);
  // initialize SX1280 with default settings
  Serial.print(F("[SX1281] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }
}
```

## Output
```
[SX1281] Initializing ... success!
18:11:25.951 -> ESP-ROM:esp32c6-20220919
18:11:25.951 -> Build:Sep 19 2022
18:11:25.951 -> rst:0x1 (POWERON),boot:0xe (SPI_FAST_FLASH_BOOT)
18:11:25.951 -> SPIWP:0xee
18:11:25.951 -> mode:DIO, clock div:2
18:11:25.951 -> load:0x4086c410,len:0xcf8
18:11:25.986 -> load:0x4086e610,len:0x2e44
18:11:25.986 -> load:0x40875728,len:0x113c
18:11:25.986 -> entry 0x4086c410
18:11:26.056 -> [SX1281] Initializing ... success!
18:11:26.088 -> �ESP-ROM:esp32c6-20220919
```
Any help will be appreciated. TIA

Thanks,
Addy

jgromes, 2025-06-22:
Since the reset seems to happen after the ESP32 prints out "success!", then I would guess the cause is whatever happens after that, which is not shown in your code. Can you post the whole thing?

adiiot-1316, 2025-06-23:
@jgromes : Thanks for responding. Below is the code. Currently i am not doing anything other than initializing.. It's just rebooting on its own. 
```

// include the library
#include <RadioLib.h>
#include <SPI.h>
SX1281 radio = new Module(18, 11, 3, 10);

void setup() {
  Serial.begin(115200);
  SPI.begin(21, 20, 19, 18);
  // initialize SX1280 with default settings
  Serial.print(F("[SX1281] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true) { delay(10); }
  }  
}

// counter to keep track of transmitted packets
int count = 0;

void loop() {
  Serial.println("In loop");  
  delay(2000);  
  return;
}
```

Sometime there is error  as part of radio.begin() resulting in errorcode = -20. 
1.)Is the wiring sequence okay? 
2.)I've not attached any GPIO pins to DIO2, DIO3, Is this fine?
3.)Any way to debug this?

Thanks,
Aditya

jgromes, 2025-06-23:
Couple of thoughts about your setup:

1. The moduel you are using seems to have an external power amplifier, as the manufacturer declares 27 dBm output power. That is much more than SX128x can do on its own (13 dBm). This external PA will draw a lot of power, and powering it just from the ESP32 3.3V line might lead to voltage drops, and reset the ESP32. Instability (such as radio sometimes reutnring an error code) is a common symptom of this, you should have an external power source for the radio.
2. You have set some pins to `SPI.begin`, but `Module` is still using the default SPI. Please check [this wiki page](https://github.com/jgromes/RadioLib/wiki/Basics#non-standard-spi-setup) on how to correctly configure non-default SPI.
3. DIO2 and 3 are not necessary, it is fine to leave the unconnected.
4. You can debug further by following [this wiki page](https://github.com/jgromes/RadioLib/wiki/Debug-mode). Enable SPI debug and see whether there are any issues during the initialization.