jgromes/RadioLib #1538: RadioLib/LoRaWan 7.2.0: FCNT on downlink goes 0-0-1-2-... downlink data only valid from FCNt 2 on

mschorer, 2025-06-24:
I have a ESP32 firmware running on a Helltec epaper, that did work flawlessly under RadioLib 7.1.2.
The epaper uplinks local node stats and receives remote nodes stats for display.

**Describe the bug**
After updating to 7.2.0 
- i get a 0-0-1-2-... sequence on downlink FCNTs
- lenDown on sendReceive gives the correct length from the 2nd FCNT 0 on, but
- correct data is only passed after FCNT >= 2

**Expected behavior**
Deliver consistent downlink data/length.

**Additional info (please complete):**
 - MCU: heltec epaper ESP32
 - LoRaWAN_ESP32 for LoRa persistence
 - SX1262
 - Arduino IDE version  2.3.6

lindenaar, 2025-06-29:
I am running into the same issues with RadioLib 7.20 (just started to play with download data at that version).

I am manually sending the data from the TTN console, tried both with and without a message formatter but that did not make a difference.

I cannot yet confirm the pattern that @mschorer sees w.r.t. to the fCnt. In my case data gets garbled about 50% of the time, it seems kind of random in my case though fCnt 2 is very often correct and fCnt 1 is always wrong, the higher numbers vary.

I tried with a 2 byte downlink message as well as with an increasing messages on each iteration, the data length is always correct and this does not seem to make.a difference w.r.t. which messages are garbled.

Downgrading to 7.1.2 immediately resolved the issue without any code changes.

let me know in case I can help with debugging this, I run my own gateway, currently connected to TTN but would not mind setting it up with something else if that could help debugging this.

StevenCellist, 2025-06-29:
Hi @lindenaar, if you can provide your code and some logs (including a bit of context when you scheduled a downlink of what size), that would be very useful!

StevenCellist, 2025-06-29:
It would be very useful to get the logs with DEBUG_PROTOCOL flag enabled in the BuildOptUser.h file. Then we can have a peek under the hood at what is received and processed. It may not provide definitive answers but will surely help narrowing down the problem.

lindenaar, 2025-06-29:
@StevenCellist sure, please find attached the output, this time none of the messages was correct content-wise (same test messages were sent).

[RadioLib_7.2.0_incorrect_messages.log](https://github.com/user-attachments/files/20969373/RadioLib_7.2.0_incorrect_messages.log)

StevenCellist, 2025-06-29:
Thank you for the logs. Since the payload is encrypted I can't be completely sure, but I suspect a regression happened while working on the MAC payload logic and/or application package logic. 
Right now, I'm wrapping up pre-certification of fixed channelplans. Hopefully tomorrow I can verify the above and deploy a fix.

StevenCellist, 2025-06-30:
I introduced an issue while refactoring MAC handling, resulting in the code looking at an incorrect index for the application payload when MAC payload is present. This is resolved in the referenced commit.

closed 2025-06-30 by StevenCellist

StevenCellist, 2025-06-30:
Thanks for the report!

commit 227e40b8 (2025-06-30): [LoRaWAN] Fix dynamic default channels, fix #1538