jgromes/RadioLib #1561: Missing error checking in SX126x::startReceiveCommon ?

MarusGradinaru, 2025-07-19:
https://github.com/jgromes/RadioLib/blob/master/src/modules/SX126x/SX126x.cpp#L618

Here shouldn't follow an `RADIOLIB_ASSERT(state);` to check the `clearIrqStatus();` result ?

jgromes, 2025-07-20:
Indeed it should, a minor oversight - feel free to submit a PR if you'd like.

MarusGradinaru, 2025-07-20:
I don't know what is a "PR"... It's my first time when I report a bug on GitHub. I don't know how things are going around here... Can you just add the missing line in the source file, so that any of us who download the library enjoy the bug free version ? If not, tell me what I need to do...

closed 2025-07-20 by jgromes

lyusupov, 2025-07-21:
@jgromes  Should you apply the ASSERT to this line as well ?

https://github.com/jgromes/RadioLib/blob/d05dbf07d12cb3a953bc13238f97ebc271b4f517/src/modules/SX126x/SX126x.cpp#L666

jgromes, 2025-07-21:
@lyusupov there it is intentionally left out, as the following line only checks whether CRC check failed. If it did not, the return value of the `clearIrqStatus` is returned. I admint it could be explained in a comment ;)