jgromes/RadioLib #1562: Radiolob and lwip wifi on Pico 2 W

ole12345, 2025-07-21:
Hi,
I am trying to use a pico 2 w to receive a message over lora radio and forward it using wifi (mqtt).

I already posted on the [raspberry pi forum](https://forums.raspberrypi.com/viewtopic.php?t=390156), but I now feel like it might be better suited here.

Radiolib is doing it's job and I can receive messages both blocking and interrupt-driven.
The issue is that I can't seem to mix the lwip wifi lib with an interrupt-driven receive.
As soon as I try to attach an IRQ to any pin after wifi is initialised and radio.begin(...) has been called the pico freezes.

Maybe you have a clue?

Here is the code example:

`#include <stdio.h>
#include "pico/stdlib.h"
#include "pico/cyw43_arch.h"
#include "lora.h"

#define IRQ_GPIO 19

void gpio_callback(uint gpio, uint32_t events) {
    // Check for the correct GPIO and a falling edge event.
    if (gpio == IRQ_GPIO && (events & GPIO_IRQ_EDGE_RISE)) {
        printf("IRQ on GPIO %u\n",IRQ_GPIO);
        gpio_acknowledge_irq(gpio, events);
    }
}

void setup_irq()
{
    gpio_init(IRQ_GPIO);
    gpio_set_dir(IRQ_GPIO, GPIO_IN);
    gpio_pull_down(IRQ_GPIO);
    gpio_set_irq_enabled_with_callback(
        IRQ_GPIO,                       
        GPIO_IRQ_EDGE_RISE,        
        true,                      
        &gpio_callback             
    );
}

int main()
{
    stdio_init_all();

    printf("Connecting to Wi-Fi...\n");
    cyw43_arch_init();
    cyw43_arch_enable_sta_mode();  
    cyw43_arch_wifi_connect_timeout_ms(WIFI_SSID, WIFI_PWD, CYW43_AUTH_WPA2_AES_PSK, 30000);

    printf("lora_init\n");
    lora_init();

    printf("setup_irq\n");
    //attach an IRQ to a free pin for testing...it will freeze here
    setup_irq();
    printf("Entering while loop\n");
    while (true) {
        printf("Hello, world!\n");
        sleep_ms(1000);
    }
}`

`#include <pico/stdlib.h>
#include <RadioLib.h>
#include "hal/RPiPico/PicoHal.h"
#include "lora.h"

#define SPI_PORT spi1  // Pico spi
#define RADIO_RESET 15 // Reset
#define RADIO_MOSI 11  // SPI MOSI
#define RADIO_MISO 12  // SPI MISO
#define RADIO_SCLK 10  // SPI Clock
#define RADIO_NSS 3    // Chip select
#define RADIO_BUSY 2   // Busy
#define RADIO_DIO_1 20 // IRQ pin

static PicoHal hal(SPI_PORT, RADIO_MISO, RADIO_MOSI, RADIO_SCLK);
static Module module(&hal, RADIO_NSS, RADIO_DIO_1, RADIO_RESET, RADIO_BUSY);
static SX1262 radio(&module);

extern "C"
{
  void lora_init()
  {
    printf("Initialising Lora...");
    int state = radio.begin(
        868.0F, //frequency
        125.0F, //bandwidth
        9U,     //spreading factor
        5U,     //coding rate
        0x12,   //sync word
        10,     //power
        8U,     //preamble length
        1.6F,   //tcxo voltage
        false   //don't use LDO
      );
    if (state != RADIOLIB_ERR_NONE)
    {
      printf("failed, code %d\n", state);
    }
    printf("success!\n");
  }
}`

`#ifndef __LORA_H__
#define __LORA_H__

#ifdef __cplusplus
extern "C" {
#endif

void lora_init();

#ifdef __cplusplus
}
#endif

#endif // __LORA_H__`

jgromes, 2025-07-21:
I don't have much experience with the Pico, so the only thing off the top of my head I can think of is that the RadioLib Pico hardware abstraction layer prforms basic initialization, namely calling `stdio_init_all()` as well as `gpio_init`, so the first thing I would try is to remove the `setup_irq` method.

jgromes, 2025-08-07:
Closing as inactive, and unlikely to be an issue directly in RadioLib.

closed 2025-08-07 by jgromes