jgromes/RadioLib #1567: Try all uplink channels during OTAA for US915 and AU915

gonzabrusco, 2025-08-07:
For LoRaWAN regions US915 and AU915, the device should attempt OTAA joins by rotating through all 64 (125 kHz) uplink channels (0–63) using DR0 and all 8 (500 kHz) uplink channels using DR4, as specified in the LoRaWAN Regional Parameters document. This is especially important for devices with no preconfigured subband, or when deploying on networks with unknown or dynamic subband configurations.

According to RP 1.0.2: For rapid network acquisition in mixed gateway channel plan environments, the device SHOULD follow a random channel selection sequence which efficiently probes the octet groups of eight 125 kHz channels followed by probing one 500 kHz channel each pass. Each consecutive pass SHOULD NOT select a channel that was used in a previous pass, until a Join-request is transmitted on every channel, after which the entire process can restart.

After a successful join the device should wait for LinkADRReq MAC commands from the network server to update the channel mask (i.e., reduce active uplink channels to the selected subband or custom mask).

Is that possible with RadioLib? 

[image]

StevenCellist, 2025-08-07:
I see that this is not really mentioned / documented: selecting subband 0 will replicate this behaviour. With the note that it selects one random channel from each subband until all subbands are exhausted, after which it does another random selection: this means that it is unlikely to cover all 64+8 channels in one go, but it will most definitely cover all subbands. It is a wish for future me to implement a deterministic selection to cover all channels without random selection. That is likely postponed until I can get my hands on a real 64ch gateway :)

gonzabrusco, 2025-08-07:
Great to hear that! I had assumed that behavior wasn’t implemented because, in all the examples, the comments mention that for US915 and AU915, we should use subband = 2. That left me unsure whether the full channel-scan behavior was actually supported, and whether the device correctly handled the LinkADRReq with the channel mask.

StevenCellist, 2025-08-07:
By all means, please try it and see how it behaves (you might want to enable DEBUG_PROTOCOL to see more info). It's a feature that isn't used much and I only try my US915 gateway once every few months as law requires me to limit transmissions to just my room only.. ;)
But as far as I know, it all fully works.

HeadBoffin, 2025-08-08:
> Is that possible with RadioLib?

Absolutely. The question is, is that a useful function as almost all the gateways in the world are 8 channel so trying 64 channels upon deployment gives the device a much longer join time. Most LNS's suggest a sub-set, hence the channel mask.

> This is especially important for devices with no preconfigured subband

Which ones are those? And can't they be configured before deployment?

>, or when deploying on networks with unknown or dynamic subband configurations.

Which ones are those? Using unknown networks has so many downsides - like how often they are available and/or move sub-band (which would render all the devices joined via it offline, so not likely to be a thing).

If you are hoping that LW networks are like public WiFi, then that's not the case.

But the age old question is the why. What is it you want to achieve? Do you want to run 8 low cost gateways in parallel or 4 expensive gateways in parallel (those that can do 16 channels)? Or really just unwrap a device, register it's credentials on an LNS and then send it out in to the world hoping that the gateway that hears it is has some sort of connection back to the LNS it is registered on? Because if you've registered it on an LNS, like you have to, then you know what channel plans are in use ...