jgromes/RadioLib #1575: Malformed JoinAccept when trying to connect to the Helium network with a Core1262-HF module.

Jonaswinz, 2025-08-15:
I'm having issues getting the LoRaWAN starter example (OTAA) to work with the Waveshare Core1262-HF (formerly Core1262-868M), a Black Pill STM32F401 MCU, and Meteoscientific Shirp Stack (Helium).

ShirpStack successfully receives the JoinRequest:

`{
    "phy_payload": {
        "mhdr": {
            "m_type": "JoinRequest",
            "major": "LoRaWANR1"
        },
        "mic": [
            175,
            194,
            89,
            143
        ],
        "payload": {
            "dev_eui": "...",
            "dev_nonce": 0,
            "join_eui": "..."
        }
    },
    "rx_info": [
        {
            "context": "+se2Sw==",
            "crcStatus": "CRC_OK",
            "gatewayId": "...",
            "gwTime": "2025-08-15T13:37:17.348+00:00",
            "metadata": {
                "gateway_h3index": "...",
                "gateway_id": "...",
                "gateway_lat": "...",
                "gateway_long": "...",
                "gateway_name": "...",
                "network": "helium_iot",
                "regi": "EU868"
            },
            "nsTime": "2025-08-15T13:37:17.363902302+00:00",
            "rssi": -115,
            "snr": -7.199999809265137,
            "uplinkId": 10637
        },
        ... Other gateways
    ],
    "tx_info": {
        "frequency": 868300000,
        "modulation": {
            "lora": {
                "bandwidth": 125000,
                "codeRate": "CR_4_5",
                "spreadingFactor": 9
            }
        }
    }
}`

and sends a JoinAccept:

`{
    "phy_payload": {
        "mhdr": {
            "m_type": "JoinAccept",
            "major": "LoRaWANR1"
        },
        "mic": [
            63,
            22,
            135,
            28
        ],
        "payload": "..."
    },
    "tx_info": {
        "context": "SJfO1A==",
        "frequency": 868300000,
        "modulation": {
            "lora": {
                "bandwidth": 125000,
                "codeRate": "CR_4_5",
                "polarizationInversion": true,
                "spreadingFactor": 9
            }
        },
        "power": 16,
        "timing": {
            "delay": {
                "delay": "5s"
            }
        }
    }
}`

but this is never received by my setup. I am always getting either a -1102 or -1116 error. I already tried different RadioLib versions and checked if there is a clock drift. In ChirpStack I also tried different MAC versions and regional parameters revisions.

This is my code:
```
const LoRaWANBand_t Region = EU868;
const uint8_t subBand = 0;

uint64_t joinEUI =   RADIOLIB_LORAWAN_JOIN_EUI;
uint64_t devEUI  =   RADIOLIB_LORAWAN_DEV_EUI;
uint8_t appKey[] = { RADIOLIB_LORAWAN_APP_KEY };
uint8_t nwkKey[] = { RADIOLIB_LORAWAN_NWK_KEY };

SX1262 radio = new Module(PA4, PB0, PB10, PB15);
LoRaWANNode node(&radio, &Region, subBand);

static const uint32_t rfswitch_pins[] = { PB2, PB1, RADIOLIB_NC, RADIOLIB_NC, RADIOLIB_NC };

static const Module::RfSwitchMode_t rfswitch_table[] = {
  { Module::MODE_IDLE, { LOW,  LOW  } },
  { Module::MODE_RX,   { HIGH,  LOW } },
  { Module::MODE_TX,   { LOW, HIGH  } },
  END_OF_MODE_TABLE,
};

void setup() {
  Serial.begin(115200);
  while(!Serial);
  delay(5000);  // Give time to switch to the serial monitor
  Serial.println(F("\nSetup ... "));

  Serial.println(F("Initialise the radio"));

  radio.setTCXO(1.8);
  radio.setRfSwitchTable(rfswitch_pins, rfswitch_table);

  int16_t state = radio.begin();
  debug(state != RADIOLIB_ERR_NONE, F("Initialise radio failed"), state, true);

  // Setup the OTAA session information
  state = node.beginOTAA(joinEUI, devEUI, nwkKey, appKey);
  debug(state != RADIOLIB_ERR_NONE, F("Initialise node failed"), state, true);

  //Serial.print("setRx2Dr -> "); Serial.println(node.setRx2Dr(0));

  Serial.println(F("Join ('login') the LoRaWAN Network"));
  state = node.activateOTAA();
  debug(state != RADIOLIB_LORAWAN_NEW_SESSION, F("Join failed"), state, true);

  Serial.println(F("Ready!\n"));
}

...
```
And this is one example log
```
Setup ... 
Initialise the radio
[12.671202] RLB_DBG: 
RadioLib Info
Version:  "7.2.1.0"
Platform: "Arduino STM32 (official)"
"Aug 15 2025" "16:03:30"
[12.673732] RLB_DBG: -707 at c:\Users\jonas\Documents\Arduino\libraries\RadioLib\src\Module.cpp:279
[12.683226] RLB_DBG: Found SX126x: RADIOLIB_SX126X_REG_VERSION_STRING:
[12.683672] RLB_DBG: 00000320: 53 58 31 32 36 31 20 56 32 44 20 32 44 30 32 00  SX1261 V2D 2D02.
[12.683813] RLB_DBG: 
[12.683868] RLB_DBG: M	SX126x
[12.685747] RLB_DBG: -707 at c:\Users\jonas\Documents\Arduino\libraries\RadioLib\src\Module.cpp:279
Join ('login') the LoRaWAN Network
[12.717864] RLB_PRO: UL:   0 1  (0 - 5) | DL:   0 1  (0 - 5)
i[12.718153] RLB_PRO: UL:   1 1  (0 - 5) | DL:   1 1  (0 - 5)
i[12.718414] RLB_PRO: UL:   2 1  (0 - 5) | DL:   2 1  (0 - 5)
i[12.718681] RLB_PRO: [MAC] 0x03
[12.718825] RLB_PRO: 00000000: 30                                               0                 
[12.718953] RLB_PRO: LinkAdrReq: dataRate = 3, txSteps = 0, nbTrans = 0
[12.719242] RLB_PRO: LinkAdrAns: 07
[12.719315] RLB_PRO: [MAC] 0x04
[12.719418] RLB_PRO: 00000000: 07                                               .                 
[12.719645] RLB_PRO: DutyCycleReq: max duty cycle = 1/2^7
[12.719739] RLB_PRO: [MAC] 0x05
[12.719968] RLB_PRO: 00000000: 00 d2 ad 84                                      ....              
[12.720122] RLB_PRO: RXParamSetupReq: Rx1DrOffset = 0, rx2DataRate = 0, freq = 
[12.727391] RLB_PRO: [MAC] 0x08
[12.727516] RLB_PRO: 00000000: 01                                               .                 
[12.727637] RLB_PRO: RXTimingSetupReq: delay = 1 sec
[12.727893] RLB_PRO: [MAC] 0x09
[12.728036] RLB_PRO: 00000000: 05                                               .                 
[12.728192] RLB_PRO: [MAC] 0x0c
[12.728324] RLB_PRO: 00000000: 65                                               e                 
[12.728468] RLB_PRO: ADRParamSetupReq: limitExp = 6, delayExp = 5
[12.728606] RLB_PRO: [MAC] 0x0f
[12.728730] RLB_PRO: 00000000: fa                                               .                 
[12.728879] RLB_PRO: RejoinParamSetupReq: maxTime = 15, maxCount = 10
[12.796113] RLB_PRO: JoinRequest (DevNonce = 0):
[12.796526] RLB_PRO: 00000000: 00 c9 6e 32 7b 61 6d 1d 8c 6e ce b8 c7 f4 95 5b  ..n2{am..n.....[
[12.796857] RLB_PRO: 00000010: 1d 00 00 af c2 59 8f                             .....Y.           
[12.797320] RLB_PRO: 
[12.797374] RLB_PRO: PHY:  Frequency =  MHz, TX = 16 dBm
[12.798429] RLB_PRO: LoRa: SF = 9, BW =  kHz, CR = 4/5, IQ: U
[13.014672] RLB_PRO: Uplink sent <-- Rx Delay start
[13.015026] RLB_PRO: 
[13.015081] RLB_PRO: PHY:  Frequency =  MHz, TX = 16 dBm
[13.016151] RLB_PRO: LoRa: SF = 9, BW =  kHz, CR = 4/5, IQ: D
[18.008304] RLB_PRO: Opened Rx1 window (115 ms timeout)... <-- Rx Delay end 
[18.121002] RLB_PRO: Closing Rx1 window
[18.124131] RLB_DBG: -705 at c:\Users\jonas\Documents\Arduino\libraries\RadioLib\src\Module.cpp:251
[18.124373] RLB_DBG: -705 at c:\Users\jonas\Documents\Arduino\libraries\RadioLib\src\Module.cpp:279
[18.124857] RLB_PRO: 
[18.124911] RLB_PRO: PHY:  Frequency =  MHz, TX = 16 dBm
[18.125992] RLB_PRO: LoRa: SF = 12, BW =  kHz, CR = 4/5, IQ: D
r[19.008289] RLB_PRO: Opened Rx2 window (675 ms timeout)... <-- Rx Delay end 
[19.681001] RLB_PRO: Closing Rx2 window
[19.695179] RLB_DBG: -705 at c:\Users\jonas\Documents\Arduino\libraries\RadioLib\src\Module.cpp:279
[19.695422] RLB_DBG: 2 at c:\Users\jonas\Documents\Arduino\libraries\RadioLib\src\protocols\LoRaWAN\LoRaWAN.cpp:1692
[19.695720] RLB_PRO: JoinAccept reply length mismatch, expected 33B got 0B
[19.695862] RLB_DBG: -1102 at c:\Users\jonas\Documents\Arduino\libraries\RadioLib\src\protocols\LoRaWAN\LoRaWAN.cpp:946
Join failed - RADIOLIB_ERR_DOWNLINK_MALFORMED (-1102)
```                                            .                 

I would appreciate some help. Thanks in advance.

StevenCellist, 2025-08-24:
@Jonaswinz any update on your -707 error?

jgromes, 2025-08-29:
@Jonaswinz configuring the TCXO voltage this way won't work, since the module has to be initialized by `begin` prior to calling `setTCXO`. The way around this is to pass the TCXO voltage as one of the arguments to `begin`.

jgromes, 2025-08-29:
@Jonaswinz I don't think it's likely the module got damaged by setting incorrect TCXO reference voltage. According to the SX1262 datasheet, the maximum current it can deliver to the TCXO is 4 mA (see below), that seems far too low to damage anything.

[image]

jgromes, 2025-09-14:
Closing due to inactivity, feel free to reopen later.